idun-project/idun-cartridge #1: Excited and Sad

paulhocker, 2023-04-13:
First, please don't feel bad about this. I was both excited and sad to see this project. As this is something I was hoping to do for a long time.

I started a project last year called Animal Shell (https://gitlab.com/retro-coder/commodore/animal-shell) which was supposed to be a replacement for the standard command line interface for C128 using ACE, CSDOS and others as inspiration. But nothing as complicated as a operating system. I have actually made some pretty good progress, at least I think so.

You have taken this idea to the next level by incorporating the PI which is also something I had hoped to do so that one day I could use python to run programs on the C64 using the RPI as the interface.

In all, I am very excited to try this out and give my feedback and maybe one day contribute.

I saw by looking at the source code that you have some inspiration from ACE as well. IMHO it would have been great if it had continued.

At any rate, I am mostly here to say that I want to help.

Out of curiosity, how long have you been working on this and have you shared it before? I have seen nothing on this before.

Will be ordering the hardware kit soon.

Paul

idun-project, 2023-04-13:
Hi Paul.

Firstly, thank you for the kind words and support. I also understand the mixed emotions, as I've had similar experiences myself.  This project started in 2020. It was "baking" for quite a while, and even paused most of last year. For better or worse, it is not in my nature to release a project until at least I find it sufficiently useful and complete (like something worthy of calling "v1.0"). 

I am thrilled that you are thinking about contributing. ACE-128 is largely preserved in the idun kernel - please see [apiref.md](https://github.com/idun-project/idun-cartridge/blob/main/doc/apiref.md). The whole system is easily extended with additional shell "tools" (a.k.a external commands). And, there's certainly room for alternative shells, too.

I wonder if you mind that I'd like to retain this "Issue" as a Feature Request, with your permission. Right now, the shell ("dos.app" is the name of the binary) is hard-coded. It should probably be a configuration variable easily changeable by the user - that way typing "dos" at the BASIC prompt drops you into the shell of your choice. With that option, plus the addition of external commands, the shell can take on differing "personalities", rather than only the MS-DOS inspired "dos.app".

If you agree, let's rename this as "Feature Request: configurable shell".

paulhocker, 2023-04-13:
Of course. Please change the title.

The history of our projects is very similar. I started experimenting in 2020 as well. 

I am glad you have decided to share this now, and I can't wait to try it out.

Am I right in saying there are not yet any binary distributions? You need to build everything yourself?

Not a problem, really, just wanted to be sure.

Your work on the VICE emulator is also particularly interesting as not many hardware projects consider it. Great idea.

idun-project, 2023-04-13:
There is a link in the README (see "Quick Start") for downloading an image to copy to microSD and insert into a Raspberry Pi. It has everything needed to run the cartridge.

idun-project, 2023-08-04:
Release v1.1.0 has a configurable shell app for both normal and alternate modes.

closed 2023-08-04 by idun-project