idun-project/idun-cartridge #2: Recommended 3D cartridge design

CraigBladow, 2023-04-29:
Thanks for the project.  I've done Propeller based projects and it is a fascinating hardware architecture.  I have the hardware kit and it is progressing towards completion. Can you recommended  or reference the cartridge design you used for 3D printing?  

Thanks,
Craig

idun-project, 2023-04-29:
Hi Craig. My own case hacks thus far have been crude, and have not relied on any 3D printing or CAD files. I start with a standard ROM cartridge case, remove what is in the way using a small saw and file, and glue the pieces together. This has been effective for one-offs, but the project _obviously_ would benefit from a better, custom case design.

I'll keep this issue open for further discussion.

Barkfin, 2023-08-27:
This is a magnificent project! Hey I can do a 3d-printed cartridge design, no problem.
I just need some geometry info of the bare board & can whip that together easily.

idun-project, 2023-08-27:
@Barkfin That's great! I'd love to see some better case options become available.

The main PCB is standard C128 cartridge size at 84x57mm. But then the top 28mm is overlapped by a Raspberry Pi Zero 2 that is sandwiched to the main PCB and is just slightly wider. On the pictures you see in the README. you can see I just leave the top portion exposed so that all the ports on the Pi are accessible. Of course, a custom case could have cutouts for ports, and the button and switch on the main PCB. This would nicely enclose everything, but obviously would only accommodate the Pi Zero, and not the larger Model B Pi's, for example. The "open top" version has the benefit of working with more Pi's.

Anyway, I think the summary is that it needs to be a case that fits the Raspberry Pi the user intends to use (RPi Zero 2 is "recommended"), but has an "extension" of 55x57mm for the bottom part of the main PCB, and the proper height dimension to fit in the Commodore cartridge slot opening.

Barkfin, 2023-08-27:
Yes, as I figured, somebody already did this:
https://grabcad.com/library/c64-cartridge-case-1

Anyway this project is completely accessible, have a look at some of the geometry details.
[image]
[image]

idun-project, 2023-08-27:
@Barkfin Yes, that is indeed the "standard" Commodore cartridge dimensions. It will fit the idun-cartridge WITHOUT the Raspberry Pi, as long as the center screw post is removed - there is no hole in the idun-cartridge PCB to accommodate that. So, the necessary modification is hacking off 29mm from the top and (if desired) reforming that area to fit over the Pi nicely.

Barkfin, 2023-08-28:
Whoops, I just realized I had the Pi-2 upside down 😄

So the Pi-2 is actually 30mm x 65mm wide (at least according to the geometrical model I found).
The cartridge model extends 2.45mm past the end of the board, so that implies a 32.5mm cut top the top case.
The Pi-2 standoffs are what, about 8mm tall?

If I just do a simple trim to the top case (plus eliminate the center post) then we get this:
[image]
(Anybody who cares to try, here are the STEP-214 files)
[Idun cartridge case.zip](https://github.com/idun-project/idun-cartridge/files/12450316/Idun.cartridge.case.zip)

idun-project, 2024-02-10:
There's a new and improved 3D case design, generously contributed to the project by @_.cygnus._ . If you want a case, then checkout the design files in the GitHub repo:

https://github.com/idun-project/idun-cartridge/tree/main/hardware/3d-printed-case

closed 2024-02-10 by idun-project