idun-project/idun-cartridge #3: Instructions For Assembling Cartridge

paulhocker, 2023-07-02:
I am finally getting around to assembling my cartridge but I was not able to find any instructions. Actually I don't even remember if it came any. Is there a page on here that goes through that?

BrianHoldsworth, 2023-07-03:
Hi. Yes, there are assembly instructions included in the package. If they are misplaced, then you can also find instructions [here.](https://docs.google.com/document/d/1oX-TJ9p55PMkxa8TxwvNVYPPENycledUnv5Y5LcuP5I/edit?usp=sharing)

idun-project, 2023-07-03:
Hi. Yes, there are assembly instructions included in the package. If they are misplaced, then you can also find instructions [here.](https://docs.google.com/document/d/1oX-TJ9p55PMkxa8TxwvNVYPPENycledUnv5Y5LcuP5I/edit?usp=sharing)

closed 2023-08-14 by idun-project