idun-project/idun-cartridge #4: Boot issues on Neo128 board with Saruman SRAM

insanity213, 2023-07-09:
The RasPi boots to arch (watching via HDMI) but I crash to the monitor when the Neo128 fires up. On initial boot it always dumps at A803B with processor state SR:31 AC:04 XR:03 YR:00 SP:F5. If I reset with the Idun’s cart button it’ll crash to the monitor at seemingly random vectors or sometimes just hangs. Disassembling memory in these areas doesn’t seem to yield anything significant, seems to just be random junk in there but admittedly I’m not the brighest when it comes to 6502 assembly. The only noteworthy difference I can think of is that I’m using the Saruman SRAM module on this Neo build.

Some things I’ve tried so far:
Powering the PiZero2 either internally or externally
Wrote a new micro SD card with the latest image
Tried the original CBM kernal and JiffyDOS
40 or 80 column mode
Reflowed IC solder joints
Retested on original C128 board with DRAM and the cart functions perfectly

I did successfully boot one time but as soon as I issued the DOS command it crashed to the monitor.
I’m starting to wonder if this could possibly be a RAM timing issue with the static ram replacement module I’m using instead of DRAM? I was wondering if you’ve used this cart with a SRAM replacement or ran into any tight timing issues when developing the cart. 

Monitor output of A8007 and A8017 looks correct.

[image]
[image]

idun-project, 2023-07-09:
From the monitor, it would also be useful to see the contents of $C1:
M C1 -> should be "00 13".

If that is Ok, then the next thing to look at is the disassembly at $1300.
D 1300 -> first instruction should be JMP $1305, followed by lots of valid 6502 code starting from $1305.

If this is no good, then the issue is on the I/O side, for sure, and not on the ROM side, of the cartridge functionality. Naturally, the timing is critical for reading the I/O at the cartridge port. It's not yet registering with me how the SRAM mod could effect the bus timing when reading from external I/O at $DE00. It would have to be effecting something fundamental with the C128 Phi2 clock OR with the "clock stretching" that it does when running in 2MHz mode.

insanity213, 2023-07-10:
I tried 40 column mode.  It seems the Idun is more likely to boot on a cold start in 40 column mode but crashes within a few minutes.  I was actually able to enter DOS then hit F7 to get to the Linux shell on one occassion.

After crashing back to the monitor I looked at the output of $C1 and $1300.  C1 was 00 00.  $1300 looks different each time I've rebooted/crashed/dumped the RAM in the monitor.  Also had one crash where it just hung on a garbled screen.
[image]
[image]
[image]
[image]

Anecdotal / mostly unrelated - In the past on the C64 I had problems with combinations of SRAM/Cartridge/Kawari VIC-II that I was able to work around by 'tuning' PHI2 a bit.  I disconnected PHI2's output from the CPU and generated my own PHI2 through a pair of inverter gates on an LS14 and manipulated it a bit with a low value capacitor in order to get the system stable.   I didn't have any of these problems in machines with DRAM.  The Kawari developer also made some tweaks to RAM timing to get his device to work with SRAM but I'm not 100% sure exactly what he did.  I think it had something to do with RAS/CAS.  

I know the 8502 in the 128 doesn't generate the PHI2 and I'm comparing apples to hand grenades but I though it was worth mentioning.  I'm wondering if CAS or RAS falls at a different rate when attached to an SRAM module and it nudges the timings off the mark just a hair, similar to the Kawari issue. Playing with the edge of PHI2 may help those timings hit the mark?  All guesses on my part really...  I wish I had a logic analyzer to dig in deeper, but I only have a cheap 2 channel scope.

idun-project, 2023-07-10:
The plot thickens. If the system goes from crashing immediately, to working but unreliabe, just by switching to 40 column / 1 MHz mode, then it makes sense to assume that Phi2 pulses are affected by the SRAM. And that is causing the cartridge IO to fail outright in 2 MHz mode and to become unreliable in 1 MHz mode.

Anything 100% reproducible that leads to a crash in 40 column mode could help localize the problem. I think 80 column mode will remain "hopelessly" broken until the root cause is better diagnosed. Any O-scope output of Phi2 at the cartridge port on a DRAM vs. SRAM C128 would also be useful.

I have to say the timing tolerances are quite tight already. If the Phi2 edge pulses are coming quicker than normal, we can't just patch the cartridge software to allow for that - spare cycles aren't available here and the Phi2 clock is what is used to "trigger stuff" in the code.

insanity213, 2023-07-18:
Problem resolved, it was related to the 1mhz clock line between the VIC-IIe and the cartridge port.  Also noteworthy was that with DRAM installed the system was marginally more stable but still had issues.  At least I was able to boot into the Idun DOS shell and run a demo or two before it would crash.  With SRAM I was lucky to get it to boot at all.  My issue seems to stem from the amplitude of the 1 mhz clock generated by this particular VIC.  I swapped it out for a spare and the system started running perfectly.  I guess some VICs have stronger clock signals than others.  See attached scope pics.

[image]
[image]

closed 2023-07-18 by insanity213

insanity213, 2023-07-21:
Yeah, kind of a weird one.  I guess having just a tiny bit less amplitude on that signal was enough to make the falling edge too short for stuff to time right.  I had some odd behavior with the Ultimate II cart as well with that weak VIC in there.  

I did try putting a 330 in parallel with the 1k on that line and it didn’t make a difference.  I didn’t want to go too much lower and let the smoke out of the Propeller though.  Thanks for your help, I really appreciate it!

Ray

From: idun-project
Sent: Wednesday, 19 July 2023 4:13 am
To: idun-project/idun-cartridge
Cc: insanity213; State change
Subject: Re: [idun-project/idun-cartridge] Boot issues on Neo128 board withSaruman SRAM (Issue #4)

Wow! Glad you solved the issue.
I see the 0.40V difference in clock amplitude. Now I'm wondering if a slightly lower value for R15 on the idun-cartridge would also allow it to work with your original VIC-IIe chip. The 1Kohm R15 (on Phi2 input) is there to not over-power the Propeller input.
If I can get my hands on a VIC-II that outputs a lower voltage clock signal, then I could do some experimenting.

idun-project, 2023-08-14:
@insanity213 I was wrong in my initial assumption that the read response time for the cartridge couldn't be improved (Propeller multi-core for the win...)

In light of the resolution to #5 , it might be worth testing again with your original VIC-IIe chip.

insanity213, 2023-08-15:
I reflashed my SD card from scratch with 1.1.1 and reinstalled my 'weak' VIC-IIe.  Your fix certainly improved the situation but the machine is still quirky with this VIC and the Idun cart.  I ran the cube app for about 15 minutes then started to see corruption in the video.  Upon exiting the cube app and going back to the shell I got some strange behavior with screen updates and cursor movements but the machine didn't full out crash like it used to.  I tried to reset with the button but struggled to get back to the shell.  Also, there may be another unrelated bug at play where for some reason 1.1.1 is ignoring the VDC config in idunrc.toml.  I'll open a separate issue for that.