idun-project/idun-cartridge #5: IDUN Not Loading on C128D(CR)

kd4cbm, 2023-07-21:
It appears as though I either have an issue somewhere or the internal drive may be causing issues with IDUN loading on a C128DCR.

RPi software wise, the OS and everything else on the SD card is bone stock with zero modifications with the exception of installing the Vice package.

Has anyone had a successful load on that specific 128 type? RPi Vice appears to be working locally, albeit a bit slow.

The cart does hold up the c128 boot process until the Pi completes loading so I assume the correct lines are toggling.
I did do another SD card where I incremented the device ID's by 1 in the config file. The result was the same though.

I have a 128 flat I might be able to test with, I'll just have to figure out where I hid its power supply.

idun-project, 2023-07-21:
Hi. Much of the idun-cartridge testing was done on a 128D. I don't have a working 128D(CR) to test with right now, but can't foresee why there would be any issue. Internal floppy (is it set to IEC device #8 or #9?) should not effect boot process.

Are your symptoms anything like those reported in issue #4 ?

kd4cbm, 2023-07-21:
I've had very, very, few break conditions. What I'm seeing is either the machine boots to a ready prompt or just hangs at the Commodore copyright notice that shows just before ready does.

I've looked at issue #4 and plan on probing around with the scope over the weekend to see what I can find.

The internal drive is still set to 8. If it works on a D, I would imagine the CR variant should work as well.

I'll post back here once I start probing around and have some useful info to share.

Thanks!

kd4cbm, 2023-08-01:
Sorry for the delay on the follow up. I wanted to rule out any hardware issues before I started probing and USPS is getting slower and slower these days.

I believe I am in fact seeing the same issue as described in issue #4, they are just presenting themselves a little differently. Differently as in the 128s are either hanging while attempting to load the cartridge data into memory, or just booting normally with no code loaded. The behavior is the same between the machines. 

On the scope, there is a pretty good amount of slew and noise on the clock line which may explain why the user who reported #4 got better results with a stronger VIC. That extra .4v may have been enough to get the timings in check. I tried dropping the clock line resistor down slightly to 800ohms with no improvement. I'm a little cautious on going too low, since I've already smoked one Propeller due to my ham-fisted Pi alignment. 

I did try the VIC from my flat 128, but the results are the same regardless of the VIC, or which machine it's in. I'm not terribly surprised by this though since both VICs were manufactured within 2 weeks of each other, and their clock outputs are nearly identical to each other in terms of output. If I can get my hands on an expansion port breakout, I'd like to try intercepting the clock output at the port and buffering it with another 74LVC245 or similar directly to the MCU pin to see if it solves anything.  Or even just jumpering from a point somewhere close to the VIC directly to the input resistor on the cart...

idun-project, 2023-08-01:
Hi. So if you have a chance later, would you please provide the date codes on the two VIC-IIe chips you have? I'd like to compare with those in my test machines, and also find a VIC-IIe of my own with which I can reproduce the problem.

kd4cbm, 2023-08-01:
4585 and 4385 are the date codes on the 2 VIC-II's I've got in hand. I'd be happy to loan you my spare if it would be helpful, especially if it leads to a solution. My C128 use case makes this particular cartridge perfect for my needs once I get things up and running. 

I would prefer to facilitate that communication via e-mail if you're interested. Google's e-mail service, same username

idun-project, 2023-08-14:
Thanks to @kd4cbm for lending me one of his R5 VIC-IIe's. The following was learned:
1. The 8564R5 is unreliable with the idun-cart.
2.  Phi2 as detected at the Propeller input is identical for R5 and R6. It does not appear to be a clock problem.
3. The difference is thermal. My flat C128 booted Idun perfectly the first time with the R5 chip, and anytime that I allowed the C128 to cool completely back to ambient temperature. It becomes unstable after a short warm up time, which affects the timing tolerance for reading from the cartridge port.

Release v1.1.1 shaves 80ns of response time for cartridge reads. This is enough for it to work reliably with the R5 chip.

closed 2023-08-14 by idun-project