idun-project/idun-cartridge #6: Out of sync 80 column mode

celso, 2023-09-03:
Hi,

First of all congrats on the idun cartridge project, I just finished soldering mine, had it running in my C128DCR and it's mindblowing. Thank you.

I'm having problems running DOS in 80-column VDC mode with my C+1084S monitor (connected with the std analog RGB cable).

It boots fine at first, shows the IDUN banner, but as soon as I type and run DOS the video gets out of sync. See attached screenshots.

[image]
[image]

I can see DOS is running properly though, I can even type commands and it's fine, just doesn't show stabilized video.

I looked into idunrc.toml for options. And found: 

```
[vdc]
offset = 169
config = [27, 16]       # VDC default lines, memory size in KB
```

Changed it ,64 since afaik the DCR has 64K VRAM, but it didn't help.

Any clues on what might be happening and how to fix it?

celso, 2023-09-03:
I did try setting an interlaced mode and got good video:

```
mode vdc 0 80 47
```

[image]

Still, I tried a bunch of non-interlaced modes changing the number of lines from 20-27, and they were all out-of-sync.

celso, 2023-09-03:
On a separate note idun doesn't seem to be picking the `/home/idun/.config/idunrc.toml configuration` at all. I changed a few settings, then resetted the cartridge, even rebooted the raspberry pi, just in case, and nothing.

Is this the correct path? Do you want me to create a new issue for this?

idun-project, 2023-09-03:
Hi. I would suspect that the monitor was having trouble syncing with the default 27-line non-interlaced text display. Changing the default lines to 24 in the configuration file should work. But typing "mode vdc 0 80 24" at the C: prompt should also work.

On my CRT I can set the number of lines up to 29 non-interlaced and 50 in interlaced mode without sync issues- just at very high line counts it does start to disappear off the top and bottom of the CRT.

Have you also checked the PAL/NTSC configuration setting is correct for your C128?

idun-project, 2023-09-03:
On the idunrc.toml question, please do open a separate issue to describe exactly what you are seeing.

celso, 2023-09-03:
> Hi. I would suspect that the monitor was having trouble syncing with the default 27-line non-interlaced text display. Changing the default lines to 24 in the configuration file should work. But [...]

Changing to lines 24 lines didn't work either, video stays out-of-sync. It's weird because I get good video before I run the DOS command.

All the interlace modes work though.

My machine is PAL for sure. I did change the video>config to 0 (NTSC) just in case, and it didn't work either.

celso, 2023-09-03:
> On the idunrc.toml question, please do open a separate issue to describe exactly what you are seeing.

I found the issue. `idumm` is picking `/home/idun/idun-sys/idunrc.toml`, not `/home/idun/.config/idunrc.toml` like I assumed (because there a file there too).

idun-project, 2023-09-03:
> I found the issue. `idumm` is picking `/home/idun/idun-sys/idunrc.toml`, not `/home/idun/.config/idunrc.toml` like I assumed (because there's a file there too).

Hmmm. Could there be a syntax error in your .config/idunrc.toml file? IF so (or if it does not exist), then the cartridge will revert to using idun-sys/idunrc.toml.

Something to try: Just copy idun-sys/idunrc.toml to ~/.config. Then, edit the .config one, reset the cartridge, and see if the setting is changed.

idun-project, 2023-09-03:
Also, I will try to recreate what you are seeing using an Amiga 1080 monitor (it's similar to the 1084S). I feel like maybe the vertical sync timing could be off, and your monitor isn't liking where the top of the display starts in the dos shell.

celso, 2023-09-03:
> Hmmm. Could there be a syntax error in your .config/idunrc.toml file? IF so (or if it does not exist), then the cartridge will revert to using idun-sys/idunrc.toml.
>
> Something to try: Just copy [...]

That was it. It's now picking the config in ~/.config. Thanks.

celso, 2023-09-03:
> Also, I will try to recreate what you are seeing using an Amiga 1080 monitor (it's similar to the 1084S). I feel like maybe the vertical sync timing could be off, and your monitor isn't liking where the top of the display starts in the dos shell.

Thanks. Let me know if you want me to run some tests.

idun-project, 2023-09-03:
Everything looks perfect on my 1080 monitor.

Does the 1084S have a "Vertical Height" adjustment pot on the back? If so, I'd try giving it a tweak, if you have not already. Also, I assume you've already tried all the front panel adjustment pots, and so those are definitely not an issue.

idun-project, 2023-09-03:
Another test: In the dos shell, type "mode debug vdc 1", then use the cursor keys to see if the graphical display comes into sync.

celso, 2023-09-03:
> Everything looks perfect on my 1080 monitor.
> 
> Does the 1084S have a "Vertical Height" adjustment pot on the back? If so, I'd try giving it a tweak, if you have not already. Also, I assume you've already tried all the front panel adjustment pots, and so those are definitely not an issue.

Yeah I played with all the vid adjustment pots in the monitor, including the vertical height one in the back, didn't have any effect on the sync issue.

celso, 2023-09-03:
> Another test: In the dos shell, type "mode debug vdc 1", then use the cursor keys to see if the graphical display comes into sync.

The graphical modes do show good video, see attached pic. The issue seems to be related with the non-interlaced text mode.

[image]

celso, 2023-09-03:
Aditional information: `mode debug vdc 3` works too. All the others 2, 4, 5, 6, 7 are out of sync in my monitor.

idun-project, 2023-09-03:
Are you willing and able to re-build the source for this repo? (using acme assembler, and you can do it right on the cartridge!)

We could experiment with tweaking the settings located here: https://github.com/idun-project/idun-cartridge/blob/fda7e2bbea1d74ec53b5439baa0757f230e4b9a6/cbm/sys/acevdc.asm#L57

celso, 2023-09-05:
Quick update. I changed the VDC register $00 (Total (Clock cycles per rasterline - 1)) from 126 to 128 (because I read somewhere that 128 CCLKs per line was the correct value for PAL machines) and I got a perfect stable video. It looks good. I'll try to learn more about the VDC timings and will do more testing later this week.

[image]

idun-project, 2023-09-05:
Fantastic!

I will also do some experimenting with that value and we can decide together whether to make a patch.

Best,
Brian

idun-project, 2023-09-16:
@celso The pull request I have been testing is slightly different than your change... it sets R0 to 127 and also modifies R4 and R7, as you can see in the diff.

Can you try this fix out on your system?

celso, 2023-09-18:
I would, but now, for some reason,  idunmm started crashing on me and I can't quite understand why. Do you have any clues?

```
[idun@idunpi ~]$ sudo RUST_BACKTRACE=full IDUN_SYS_DIR=/home/idun/idun-sys /usr/bin/idunmm
Startup idunmm, mode switch = false
Attempt to DL and Run: /home/idun/idun-sys/idun.binary
Using GPIO pin 27 as Propeller reset, active LOW
Downloading file to port /dev/ttyAMA0
5780 bytes sent                  
Verifying RAM
Starting idunio...
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Nix(Sys(EACCES))', idunio/src/pipe/service.rs:242:37
stack backtrace:
   0:   0x5ba280 - std::backtrace_rs::backtrace::libunwind::trace::h41cb1fa3f8dc6fbe
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/../../backtrace/src/backtrace/libunwind.rs:93:5
   1:   0x5ba280 - std::backtrace_rs::backtrace::trace_unsynchronized::h3fb2a2486f2aac18
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
   2:   0x5ba280 - std::sys_common::backtrace::_print_fmt::h081997ec779d2f07
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/sys_common/backtrace.rs:65:5
   3:   0x5ba280 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h8da4af7a5ba9bfae
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/sys_common/backtrace.rs:44:22
   4:   0x5e08c8 - core::fmt::write::h180d14f042e54105
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/core/src/fmt/mod.rs:1213:17
   5:   0x5b5c10 - std::io::Write::write_fmt::he8310b90476b3ea7
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/io/mod.rs:1682:15
   6:   0x5ba068 - std::sys_common::backtrace::_print::h9e5fa6cf827dcad1
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/sys_common/backtrace.rs:47:5
   7:   0x5ba068 - std::sys_common::backtrace::print::h86f17b6e8848a8c3
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/sys_common/backtrace.rs:34:9
   8:   0x5bba74 - std::panicking::default_hook::{{closure}}::hb8d00e666a5521d2
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:267:22
   9:   0x5bb724 - std::panicking::default_hook::hde0dc768c22ada1f
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:286:9
  10:   0x5bc11c - std::panicking::rust_panic_with_hook::hae1c9a2bd6483f43
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:688:13
  11:   0x5bbe78 - std::panicking::begin_panic_handler::{{closure}}::h7cd37799e166d66e
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:579:13
  12:   0x5ba7a4 - std::sys_common::backtrace::__rust_end_short_backtrace::hb6401261c79ace6b
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/sys_common/backtrace.rs:137:18
  13:   0x5bbc38 - rust_begin_unwind
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:575:5
  14:   0x45e6c4 - core::panicking::panic_fmt::h1e94fdc977619ade
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/core/src/panicking.rs:64:14
  15:   0x45eaec - core::result::unwrap_failed::h4bba05bf61cb2ae4
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/core/src/result.rs:1790:5
  16:   0x46fef4 - core::ops::function::FnOnce::call_once::hab606b88dec4da67
  17:   0x4951c4 - once_cell::imp::OnceCell<T>::initialize::{{closure}}::hd4568bcebdbe5590
  18:   0x550480 - once_cell::imp::initialize_or_wait::h15311a41ec122ca6
  19:   0x4577e4 - once_cell::imp::OnceCell<T>::initialize::h035df3be63cc482c
  20:   0x46b140 - idunio::pipe::Pipe::new::hd7551fd86db915f1
  21:   0x46a578 - core::ops::function::FnOnce::call_once::h480243b170984823
  22:   0x49527c - once_cell::imp::OnceCell<T>::initialize::{{closure}}::hdf2b1c7d8028aec5
  23:   0x550480 - once_cell::imp::initialize_or_wait::h15311a41ec122ca6
  24:   0x45782c - once_cell::imp::OnceCell<T>::initialize::h469d80821aa857b9
  25:   0x46acdc - idunio::pipe::Pipes::exists::hc19ba8574426b62d
  26:   0x4623c8 - idunio::main::h3b0d7125c9f545af
  27:   0x466b08 - std::sys_common::backtrace::__rust_begin_short_backtrace::h9182911fbf5a888a
  28:   0x467114 - std::rt::lang_start::{{closure}}::h6d092c2e54a847af
  29:   0x5b0b9c - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once::h2e815933b2956fbe
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/core/src/ops/function.rs:287:13
  30:   0x5b0b9c - std::panicking::try::do_call::ha83dadc98704a743
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:483:40
  31:   0x5b0b9c - std::panicking::try::h9fbb3a7ab7ec19fe
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:447:19
  32:   0x5b0b9c - std::panic::catch_unwind::h9d06bfa7c071d5ce
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panic.rs:140:14
  33:   0x5b0b9c - std::rt::lang_start_internal::{{closure}}::h6b34896dae3a7749
[... cut ...]
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:447:19
  36:   0x5b0b9c - std::panic::catch_unwind::h11d7ff61c13b1ad6
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panic.rs:140:14
  37:   0x5b0b9c - std::rt::lang_start_internal::ha4b531ee8cc413b2
                       at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/rt.rs:148:20
  38:   0x4654e4 - main
  39: 0xb6d4d9dc - <unknown>
  40: 0xb6d4daf0 - __libc_start_main
idunio exited status Exited(101)
```

idun-project, 2023-09-18:
Are you on v1.1.2?

It could be a configuration file problem. Is it possible that both .config/idunrc.toml and idun-sys/idunrc.toml are both modified? A fix t try would be copying unmodified idun-sys/ config file over the .config/ one, then customizing whatever you need changed in .config/.

celso, 2023-09-18:
I thought about that and tried it, didn't solve the issue. I went through some arch package updates before, maybe I upgraded one of the libs that idunio links again't and triggered this problem. Can you check the source line numbers in the trace I sent here and provide more context on what idunio is trying to do? I don't think you released the sources for these binaries did you?

One other option is to burn the image again and start from scratch.

Nore sure about the version I have is, or how to check it, but I downloaded the sd card image ~2w ago. sha256sums are:

1e0ea720f50ffc179cd3635680db6b53390ef490ff506220d81be8e181d7c7e4  /home/idun/idun-sys/idunio
fa1b9e27f5b174e21e6a61e08aa6347da8090e90f228b72daa72810c4f6812b3  /usr/bin/idunmm

celso, 2023-09-18:
Nevermind, as I wrote the last msg, I thought about running idunio trough strace, and found the issue:

`openat(AT_FDCWD, "/dev/uinput", O_WRONLY|O_NONBLOCK) = -1 EACCES (Permission denied)`

Turns out that some package I installed change the /dev/uinput permissions to 0600. Changed them to 0666 and the issue went away, all working now. I'll try your PR later today.

celso, 2023-09-19:
@idun-project I tried your vdc_pal branch settings and it works well, picture is stable. I think you can merge.

closed 2023-09-21 by idun-project

mvcube, 2025-08-17:
I've just filed a new issue (#27) because the problem doesn't seem to be solved for me. I'm on the most recent version.