idun-project/idun-cartridge #9: Pi/idun cart not booting

anovaprint, 2023-10-19:
Struggling to boot into my idun cartridge and hoping for some assistance. My main questions are what order I should be setting this up in and what I should expect when booting. Also, this is the first board I've assembled on my own so I'm leaning hard towards some type of user error here. I'll try to include pictures in case I've done something horribly wrong (I've got some flux remover in the mail to clean it up a bit better)

I'm using the supplied SD card and the original pi zero w which I understand will want to be powered independently. However I simply get a series of blinking lights on the pi and I never see anything come up on screen. I also tried booting the pi separately without anything attached and get the same result. It ends up with seven blinking lights which is supposedly "kernel not found" but I couldn't tell you if that is accurate or not. Hitting the reset button seems to have no effect

I'm especially curious in what order I should be powering things on. I assume once everything is assembled and inside the 128 I would want to power on the pi first and then power up the 128. I assume I would want to give it a couple of minutes to boot and eventually I would see something on screen on my 128. Does that sound correct? Should I be able to boot the pi independently without anything attached and verify it is at least booting properly?

Also had a bit of a follow-up question on what those partitions should look like on the SD card. Is there anything extra I might need to there? 

I'm probably missing something very silly and appreciate any assistance on what I might be doing wrong here and how I can get going on this project.

Much thanks
[image]
[image]
[image]

idun-project, 2023-10-19:
Hi. I think the problem is that you are using a Pi Zero W, which is not supported with the latest Linux image. The issues is that the image is for an ARMv7 device, and the original Pi Zero is ARMv6. This will certainly prevent the RPi from booting, as the kernel image is just not compatible.

If you do not have access to another Raspberry Pi, then we can try using an older image for the RPi Zero W. You'll need to download the image from this link and flash it to the SD card using the Raspberry Pi Imager app, or similar: [idunpi_armv6.7z](https://drive.google.com/file/d/1-f4fZqi1PGydYXPUaHLZCb5UNGgmPkZW/view?usp=sharing)

If the above image boots your RPi successfully, then just give it about 10-15 seconds to fully boot, then turn on your Commodore. The Commodore will then show the Idun boot screen, if the cart is working properly.

There may be a path forward for using the original RPi Zero, but only for awhile. The reason I was forced to drop support for it is that upstream Linux providers are dropping it because the Raspberry Pi Foundation has announced it is end-of-life, I believe in 2024. Obviously, they are anxious to move users to the RPi Zero 2.

anovaprint, 2023-10-20:
Hey, I think that's it! I really appreciate you walking a first time user through it all. 

I have a couple of pi 4s that are locked down on other projects so using the older armv6 image at least tells me the cartridgeassembly seems to have gone as intended.

This will be a good excuse to order the new zero 2 w so I can continue to keep up with the project.

Thank you so much for the help!

[image]

anovaprint, 2023-10-20:
I'll let you know if I have any issues with the new pi but I expect it will go smoothly

closed 2023-10-20 by anovaprint

idun-project, 2023-10-20:
Happy to help and glad to see that was the problem.

The Pi Zero 2 is a worthwhile upgrade and seems to be the same price; so I guess that is why they are retiring the original Pi Zero.