idun-project/idun-cartridge #10: Idun Cartridge Testing Results for 2B and 4B

paulhocker, 2023-12-02:
I FINALLY got around to building my board. No issues with the build, although I was curious if the "arcade" switch was supposed to come with the kit or if I was supposed to get that myself. Not a big deal, actually.

At any rate, I did not have a Pi Zero 2 (on order, short supply) so I used a Model 4B I had spare, and a Model 2B as well. I do have a model 3B+ I will try as well.

At any rate, I had zero issues with the 2B, but the model 4 gave me all kinds of trouble. I have included 2 video links for you to view.

The first link, is me running with the Model 2B, pretty boring but of interest is that I continually kept using the C128 reset button expecting it to boot to idun. Is it supposed to, or will that only happen when using the reset button on the CART?

https://youtu.be/jbOK3jEa3vY

The next link is me using the 4B. The video shows a couple of times where running hello.app actually crashed the cartridge, and then something weird near the end I reset the cartridge and it loaded the arcade.app. I had to shutdown and reboot the cartridge to return to regular mode. 

https://youtu.be/ft0YlNeRaXc

Anyway, it is not all bad, and I hope to do some more testing over the holidays.

This is a very exciting project.

paulhocker, 2023-12-02:
I found the docs in the base repo that state most apps can be exited using the run/stop key or pressing the c= + back arrow

neither of these work for me

using run/stop on any pi results in a crash -- c= + back arrow does nothing as well

I am testing using the hello.app

idun-project, 2023-12-02:
Hi Paul. Thanks for all the feedback. A few quick thoughts before diving in deeper on any of the issues you are seeing:

- The C128's built-in reset button will not reset the cartridge. If you are in the idun shell, then it will return you to BASIC with a fully reset machine and the cartridge "invisible". This is the same as typing the "exit" command in the shell. The "big red button"  resets the cart+C128.
- I run my test machines with JiffyDOS enabled. It's not a problem.
- I'm **curious to know if you have the latest** v1.1.3 idun software installed.
- Try holding down "STOP" for a second or two to exit the .app sample programs.
- My own testing with the RPi 4 has been very limited so far. I've done a little bit with a 4B many months ago. I'm currently waiting on a Compute Module 4 with an adapter to test more vigorously, with an eye toward eventually releasing 64-bit (ARMv8) versions alongside the ARMv7 I'm doing now.

idun-project, 2023-12-02:
I see I shipped your kit back in mid-April. The little switch for the arcade mode was not included at that time because the feature came a few months later. The new kits have this part: https://www.digikey.com/en/products/detail/te-connectivity-alcoswitch-switches/1825232-1/4021554

The most mysterious of your issues is that you obviously invoked arcade mode without the switch and without modifying your idunrc configuration file. This was with the Pi 4B? It smacks of an electrical contact being made on the PCB where the switch is normally installed... acting as a "ghost switch". Something to be on the look out for if you don't install a switch.

idun-project, 2023-12-04:
Hi. I put in a fix for the hello.app freezing. It is commit aec4dd6. If you want to try the new binary, it is in idun-base/apps too.

I think we can close this issue. Thanks for the feedback.

closed 2023-12-17 by idun-project