idun-project/idun-cartridge #12: VDC mode

Djoulzy, 2024-01-05:
Hi,
I have a C128DCR with 64K VDC RAM factory installed, but I can't set mode 2,4,6 and 7, using "mode debug vdc x".
My monitor is a 1084S (come from ann Amiga set).

The 64K vdc mem is correctly detected by the memtest tool ...

Did I missed something ?
(My idun version is 1.1.4)

PS: Congrats for this fabulous piece of hard/soft ware ;)

idun-project, 2024-01-05:
Hi. It _might_ just be that the default configuration file is setup for 16K VRAM. Have you tried changing it to 64K.

The file to edit is ~/.config/idunrc.toml and the VRAM memory setting is around line number 150.

Djoulzy, 2024-01-06:
Yep ! It was exactly that !
I should have read the doc before asking ... My 1084S is certainly not the best screen for those resolutions, but it works !
Thank you (and happy new year) !

idun-project, 2024-01-06:
Excellent, and Happy New Year!

closed 2024-01-06 by idun-project