idun-project/idun-cartridge #21: would be great to have a mega 65 version of this gadget

country-bumpkin-software, 2025-05-30:
Would it be difficult to convert this gadget to accommodate the mega 65?

idun-project, 2025-05-30:
I don't have a Mega65 and have not tried connecting idun-cart to one.

I think the goal of the Mega65 team is to have it's expansion port support cartridges that work on other Commodores, at least to the extent they can manage. Based on online anecdotes, some cartridges work and others do not. So, it's more a question about compatibility.

I'll ask on the Discord - maybe there's a user with a Mega64 that can try it.

country-bumpkin-software, 2025-05-30:
Ive got a mega 65 i can maybe give it a try i dont have the gadget though thats the only thing - i might need to order one.  plugging it in to the mega65 wouldnt fry the computer ?