idun-project/idun-cartridge #27: VDC on PAL machine: Display is unstable

mvcube, 2025-08-17:
My Panasonic TV (LCD, 720p, connected to Scart via a simple passive D/A converter) has problems producing a stable image in 80 column mode when in the IDUN shell or any of the dependent environments like Linux or CPem. The screen seems to lose sync for a split second every few seconds. I assume that the VDC timing is not quite correct for PAL machines.

In BASIC, 80 column mode has no issues whatsoever.

idun-project, 2025-08-19:
I don't know the cause of the flicker, but would focus first on your RGB->SCART converter thing, since that is unfamiliar. Where could I find the one you have?

We can also test with graphics modes to see if any of them work without flickering. From the idun-shell:
```
c:> mode debug vdc 1 #displays graphical output; press STOP key to exi#28
c:> mode debug vdc 3
c:> mode debug vdc 5
```
IF you have 64KiB of VDC RAM, then you can also test interlaced modes 2,4,6,and 7.

You can test out interlaced text mode (works with 16KiB VRAM) using `mode vdc 0 80 45` to set the VDC display to 45 lines.

THEN, we can maybe check if other VDC programs are working without the flicker. Launch them from BASIC like so:

1. RESET cart
2. cd "idun-base/games" #change to games dir
3. cd ":games.d71" #mount the disk image
4. F3 to show directory
5. Scroll up and place a "%" in column 1 of "hearts" or "invaders", and press RETURN.

Neither of those games does any thing special with the VDC sync settings.

mvcube, 2025-08-24:
Here is some more info:

I've been using "glitch" and "flicker" interchangeably for a single phenomenon: the momentary loss of horizontal sync, most probably for just a single frame. This only occurs on my Panasonic flatscreen TV (on older 720p model with SCART and HDMI inputs). The Mitsubishi has no problems with the timing in text mode. All the other problems, notably the garbage display in interlace mode, occur on either screen.

I've created a small BASIC program that dumps the default VDC registers to a file. Here is the contents in PC readable form:

[VDC.ods](https://github.com/user-attachments/files/21958606/VDC.ods)

I'm pretty sure that the idun kernel writes different values to the VDC registers than what the C128 kernal does, which, in turn, is causing the display problems. This is my main concern.

mvcube, 2025-08-28:
Never mind: The option -y in pacman did the trick. Suggestion: Update the setup.sh script to include the option.

SUCCESS: The glitches seem to be gone in non interlace text mode.